Staacks/gbinterceptor #4: Does not recognize second boot on special cartridges with menus (i.e. Everdrive, Game Genie, etc.)

Staacks, 2022-12-21:
The Interceptor waits for a request to address 0x0100 to recognize the start of the game. When there is an error or the game halts too long, it shows an info screen, but will not again wait for 0x0100 unless the Game Boy is turned off and on again. While this makes sense for regular games, it interferes with some special cartridges, that show a menu before loading a game and reset without turning off.

To fix this, the Interceptor simply needs to allow for a second boot within a short time frame.

**Workaround until this is fixed:**
Plug in the USB cable for the Interceptor when the special cartridge is already showing its menu.
(This way it misses the first 0x0100 and starts with the 0x0100 from the actual game. You won't see the menu on the Interceptor, though.)

Staacks, 2022-12-29:
Uhm... github was a bit fast here. I leave it open until someone with an Everdrive reports back.

Staacks, 2022-12-30:
Confirmed to fix the problem for the Everdrive.

closed 2022-12-30 by Staacks

domis4, 2023-04-03:
have this issue with the EZ Flash JR. (besides the already known graphical issues)

can i support or help here?

[image]

Staacks, 2023-04-03:
Wait, this only happens when the EZ Flash reboots? This looks exactly like the MacOS problem on M1 and M2 macs and the original EZ Flash problem was already resolved a while ago and was simply the problem that the Interceptor did not recognize that the Game Boy restarted and was stuck on the error screen. This is a USB Video problem and I am scratching my head how it would be related to a reboot of the EZ flash. Also, I am using an EZ Flash Jr. all the time to test game-specific problems...

So, let's start collecting a few details:
- Which firmware are you using (and if not the latest, please try the latest one).
- Can you tell me step by step what you do and what the Interceptor shows? Especially, what part looks correct, which status/error screens are shown (even for a fraction of a second) and what the blue LED on the back does.
- Can you also show a Windows screenshot? I am wondering if it is the MacOS issue on Windows, too, or if we are looking at two problems at the same time.
-

domis4, 2023-04-03:
My steps where the following and reproduceable:

Windows 11:
1. Plug in Everything (GB Interceptor into GBC, Cardridge into GB Interceptor, USB C into Windows 11 PC)
(no LED)
2. Launch OBS and select a new Scene with the GB Interceptor 
(no LED)
3. Power on the Gameboy, wait until the screen is visible in OBS
(red LED, until OBS picked a signal, then both LED)
4. Launch a Game
(red LED only)

[image]
[image]
[image]
[image]

Staacks, 2023-04-03:
Thank you for your e-mail. I will be out of office until 11th April and my response may take longer than usual.

Best regards,
Sebastian Staacks

Staacks, 2023-04-03:
Thanks, so the green video problem is the known MacOS issue. (Or is the EZ Flash menu normal on the Mac?)

But I am afraid that Pokemon Crystal is not a good example, because that is a Game Boy Color game which is not supported by the Interceptor. I would guess that the error is not from the second boot but from the GBC game entering double speed mode. Do you have the problem with classic games, too?

domis4, 2023-04-05:
well, thats embarrsing. It works perfectly on .gbc games 🤦‍♀️ how could i forget that crystal is a gbc only game. Thanks for your fast replies!

commit 988e73ec (2022-12-29): Allow restart without waiting for the Game Boy to be turned off. This should fix #4, but has not yet been tested.