Staacks/gbinterceptor #6: R-Type DX

Megabusterretro, 2023-01-07:
**Type of issue**
**Interceptor reports error / Game looks different from Game Boy screen** / other

**Describe the bug**
A clear and concise description of what the bug is.
-Upon booting, graphical anomalies.
-Interceptor reports error when launching on IPS Modded GBC and IPS Modded SP after pressing start to start the game.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Host software (please complete the following information):**
 - OS: [e.g. Windows, MacOS, Linux, ...] Windows 10
 - Software: [e.g. OBS, VLC, ffmpeg, Zoom, ...] OBS

**Game Boy:**
- Model: [e.g. Game Boy Advance AGB-001]
Q5 IPS GBC
IPS Modded GBA SP (Boxypixel Unhinged)
**Game:**
- Title: [e.g. Tetris] R-Type DX
- Version: [Usually printed on the cartridge, e.g. DMG-ZL-NOE]
DMG-AWHE-USA
**Additional context**
Add any other context about the problem here.
[image]
[image]

Staacks, 2023-01-07:
Thanks for the detailed report. But isn't this a Game Boy Color game? The Interceptor only supports classical ones.

Megabusterretro, 2023-01-07:
It is a GBC game, but it it also works on the DMG/Classic one.  There were only a handful of games released like this.  It works just fine on the DMG/Classic GB.  The cart is solid black vs the transparent shell.  This thing performs amazing with the Classic GB though, Great Work!
[image]
[image]

Staacks, 2023-01-07:
Ah, ok, that makes sense. Those games should work fine when run on a DMG, Pocket or Light, but will not work on anything that supports the color mode. Maybe I could eventually make them also work on the Game Boy Color because I suspect that these games would not run in double speed mode, so it should be feasible to just implement a few features that trip the Interceptor here.

Nevertheless, I close this issue here because at the moment this is simply not supported by the Interceptor. I will also add R-Type DX to the compatibility list (I understood correctly that it works as expected on the DMG?).

closed 2023-01-07 by Staacks

Megabusterretro, 2023-01-08:
Yeah, DMG Works great!