Staacks/gbinterceptor #7: Fortified Zone

Megabusterretro, 2023-01-08:
**Type of issue**
Interceptor reports error / Game looks different from Game Boy screen / other
**Black Screen when starting main gameplay**
**Describe the bug**
A clear and concise description of what the bug is.
**Game starts without issue, able to start the game but once the first gameplay screen appears the Interceptor screen goes black.**  
**To Reproduce**
Steps to reproduce the behavior:
**1. Start Game**
2. Click on '....'
3. Scroll down to '....'
4. See error

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Host software (please complete the following information):**
 - OS: [e.g. Windows, MacOS, Linux, ...]  **Windows**
 - Software: [e.g. OBS, VLC, ffmpeg, Zoom, ...]  **OBS**

**Game Boy:**
- Model: [e.g. Game Boy Advance AGB-001] **Funnyplaying IPS modded DMG and IPS Modded GBC**

**Game:**
- Title: [e.g. Tetris] **Fortified Zone**
- Version: [Usually printed on the cartridge, e.g. DMG-ZL-NOE]  **DMG-IY-USA**

**Additional context**

Add any other context about the problem here.
[image]
[image]

Staacks, 2023-01-08:
Jepp, I can reproduce both problems: The glitch during the title screen effect does not surprise me much. The PPU emulation timing can certainly use some improvement and this looks like a good candidate to work on a few details.

In terms of the black screen I have no idea right now what is happening. I did not expect this and the fact that it remains black even if you restart the game seems like it is not just an emulation problem, but a more basic bug of the Interceptor.

Will hopefully find some time to look into it in a few days.

Megabusterretro, 2023-01-08:
Sounds good!  I have probably ~200 game boy games that I can test for functionality if you would like.  Would this be the best method of submitting a list of verified working titles?  Just submit a Game Bug issue and just state the list of titles working and a screenshot?

Staacks, 2023-01-09:
Woah, that would be amazing. Any form you can send me the list works for me.

Staacks, 2023-01-09:
Found the main issue of the black screen: I had the emulated div register linked to my internal clock counter, which stupidly meant that the clock counter jumps when a game resets the div register. This should be a problem for some other games too, so I directly turned it into a new release.

I don't think that I will look into the glitch of the title animation anytime soon, though. Most mid-frame effects work fine, but in this case I think it is not an issue of the inter-line timing, but the intra-line timing (if these terms are a thing). It looks to me as if the problem is that the scroll register is updated while the Interceptor is not yet finished with each line. Unfortunately, the intra-line timing is very tricky on the Interceptor because a) there are some situations in which it just cannot render fast enough and lags slightly behind (making up for it during hsync) and b) the USB communication on the same core can slightly delay the rendering of a line. So, this is a bigger thing to tackle at some point with a few creative ideas...

closed 2023-01-09 by Staacks