Staacks/gbinterceptor #8: Firmware 1.0.1 and 1.0.2 cause connecting video feed to crash - across MacOS, Windows and Android.

remy, 2023-01-09:
**Describe the bug**

Firmware 1.0.1 and 1.0.2 cause connecting software to crash - across MacOS, Windows and Android.

It looks like it's only when the video feed is probed / previewed that my software then goes into a crash.

Does not happen with 1.0.0 firmware though. I'd normally expect this is unique to my computer, but this happens across totally different devices.

I've also been able to replicate between multiple GB Interceptors with the same effect. The PCB print is from version 1.0.0 too.

**To Reproduce**
Steps to reproduce the behavior:

1. Install either 1.0.1 or 1.0.2 firmware on GB Interceptor
2. Try to view video stream using methods described in wiki

**Expected behavior**

Video stream is found and doesn't crash software.

**Host software (please complete the following information):**
 - OS: MacOS 11.7, Android (latest Google OS with Pixel 6a Plus), Windows Surface with Windows 11
 - Software: 
	 - On Mac: VLC and Quick Camera - using VLC and selecting Open Capture Device, VLC then hangs until it crashes
	 - On Windows: OBS but also "Manage Camera" (note that on Windows I can see the device correct detected and listed in the device manager, but once I got to Manage Camera and select GB Interceptor, it tries to get a preview and crashes the settings panel)
	 - On Android: USB Camera (com.shenyaocn.android.usbcamera) - app will crash and immediately reopen (in the background) finding the USB connection again, but bringing the app to focus then causes it to completely crash out

**Additional context**

Am testing **without** gameboy or game attached. 

I appreciate there's not a load of detail here, and I suspect this isn't happening for many/if _any_ people, so happy to add more detail where you think it would be useful.

Staacks, 2023-01-10:
ok, that is a weird one indeed, because I just tried on Windows and Mac without problems, but I can reproduce the problem on Android. And I can confirm that it is no problem with 1.0.0.
Not sure why it happens on some devices and not others, but at least I can reproduce it and have a very good starting point to narrow it down.

The strange thing is that the Apple fix in 1.0.1 of course sounds like a good candidate that influences USB behavior, but it basically only accepts a somewhat incorrect request that otherwise would have been rejected. I just hope it is one of my few changes and not something in TinyUSB which I updated for this release - then I would have to do quite a bit of diggin.

I hope I can try a few things this evening. (day job first)

remy, 2023-01-10:
I have another Windows and another Mac I can test on too - I don't particularly expect different results, but it might be worth checking on my side too.

If there's anything I can help debug or look at I'd be happy to do what I can.

remy, 2023-01-10:
Just writing to confirm - testing on the second Mac (different model and older MacOS) and the same thing happened. I tried it with VLC and when I opened "Open Capture Device", VLC crashed out.

Staacks, 2023-01-10:
Ha! First change from 1.0.0 to 1.0.1 I checked resolved the issue for the Android app: With 1.0.1 I set the allowed fine control of frame rates to 100ns intervals - effectively allowing any framerate below 29fps. If I only allow larger intervals the app works fine. These fine intervals should only offer more options to the app, but I assume that this particular app tries to enumerate all possible options and that allowing anything leads to too more options than it can handle. Anyhow, here is a test build with the fix:

[gb_interceptor_1.0.3-alpha1.uf2](https://cloud.tankwar.de/index.php/s/qJryDySDeTpiD8o)

Please check if it solves the issue on Android for you, too.

Unfortunately, I am a bit skeptical that this also resolves the other issues, because if the other apps were susceptible to this problem, I should be able to reproduce it here, too.

If it does not resolve the issue for the other apps:
- Did you try 1.0.0 on those apps, too? So, can I still assume that it is a change to 1.0.1 that broke the other apps for you?
- Could you try a different cable? I know that this sounds like one of those out-of-ideas suggestions (especially when we are only running a 1MB/s stream over a USB-C cable), but I once had a similarly weird issue that was fixed by a different cable. (Would not explain why 1.0.0 worked, though)
- Do you have a Raspberry Pi Pico? If so, you can simply upload the uf2 file to that one, too, and should be able to see the standby screen using the Pico. Again, this would not fit that 1.0.0 works, but it would be a way to rule out hardware issues with your particular Interceptor.

remy, 2023-01-10:
So, 1.0.3-alpha1 worked on Android.

Then I went back to 1.0.2 and thought about the cable. I had originally tried two different cables, but both were USBC to  USBA with an adapter back to C.

I tried a USBC to USBC cable, and that was it - it started working on the Mac. I'd be willing to be it'll work on the Windows machine too (I don't have it at hand, but I will do later on).

I can't quite think why it would make a difference, the 160x144 video stream surely can't require whatever the USBC 'many-pins' offers, but apparently it does make a difference.

I can still try directly loading to a Pico if that's useful (to you for testing), but I hadn't tried it yet.

Staacks, 2023-01-10:
No, thanks, I think the Pico test is not necessary then. I think I have to do some reading on USB at that point to understand this. I know that even though it is only 160x144, the Interceptor can still be a problem because if I understand correctly USB 3 is treated separately and the bandwidth for the slower standards is still limited, especially with isochronous transfer. But I have no clue how the cable plays into that and indeed why a cable that happily handles USB 3 could be a problem here. Maybe my design of the USB line is not ideal.

I will close this issue for now and add a note on this to the trouble shooting guide.

closed 2023-01-10 by Staacks

Staacks, 2023-01-12:
Just a quick question as I am still working on the Apple issue: Is it working on both Macs and can you tell me the models? (in particular if they are Intel-based or M1/M2 models)

remy, 2023-01-12:
I'll check the other Mac in the morning (it's my wife's which she uses for
work), but neither are M1/2 Mac's - they're the intel flavour.

Staacks, 2023-01-13:
Thanks, I just filed a bug report with Apple yesterday, referenced #1 and hope that a few more samples of working Intels vs. not working M1/M2 makes it clearer that even if there is a bug in the Interceptor there at least is an inconsistency at their and on how their devices handle it.

remy, 2023-01-13:
Completely unrelated, but I didn't want to open an issue just for this question: what does the mode button do on the PCB?

Staacks, 2023-01-13:
It toggles between gray and green colors as well as frame blending on/off. (While the game is running, firmware version 1.0.3 also shows the state of the frame blending on the screen when it changes)

LordRembo, 2023-02-01:
I've been testing firmware 1.04, with a Game Boy Camera as a webcam.
Using it with Google Meet, on an Intel MacBook Pro from 2015, with Monterey on it (macos v 12.6).
The feed as-is works flawlessly. It's just when running it through OBS and using it as a Virtual Camera that the feed starts breaking down.
In my case: previewing in OBS is fine, but running the Virtual Camera through Google Meet causes lagging (constant flashes of green screen between frames) and every now and then the camera just turns of completely. 

I don't think it's a problem, and certainly not something the firmware should solve. Nobody is going to use this setup for real and I don't know how it holds up with another service. Just thought I'd mention this here for posterity and I'm curious how other people's experiences are with it as a webcam.

Staacks, 2023-02-01:
First of all, thanks for another working data point on an Intel Mac :) If only Apple would even look at the bug report I sent them...

There are two details that might help me understand the issue:
- Are the issues only visible in Google Meet while the preview in OBS remains stable or is the breakdown also visible in the OBS preview while Google Meet is running?
- Does it help if you explicitly set the framerate in the source settings for the Interceptor in OBS to a lower value, like 25 fps?

LordRembo, 2023-02-01:
No problem! Would be nice if Apple got their act together, but alas.
- The OBS preview itself is stable, no issue there.
- The lowest frame rate I could choose in OBS was 10fps and the results are always the same. I also tried various video qualities from 1920x1080 all the way down to 1024x576 (my screen ratio is 16:9)

In Google Meet itself, I set the quality at the lowest (for the video's I'm seeing as well as what I'm streaming).
It's unfortunate Google Meet doesn't have much in the way of settings so I couldn't try more than that.

Staacks, 2023-02-01:
No, wait, now that I read it again, it is going through OBS after all...?
In that case: What happens if you use a different source in OBS. Or maybe even more interesting: Running the Interceptor in the same frame as some other source, side-by-side. I am wondering if it really is an Interceptor issue if it reaches OBS without flickering and problems only occur after the composited image is pushed through OBS's virtual webcam.

LordRembo, 2023-02-01:
Good catch, can't believe I didn't think of that. I did some more testing by 1) routing the built-in camera of the MacBook into OBS as source for the virtual webcam and 2) combining the built-in cam as well as the Interceptor. Used with Google Meet as well as an alternative conferencing service for comparison (Jitsi).
In all combinations, both conferencing services give green screens with some flashes of images/blackness when using the OBS virtual cam. So seems it has nothing to do with the GB Interceptor, it's just the online services (on Mac at least) that are terrible :D
Thanks for taking a look along!

Staacks, 2023-02-01:
Good to know. Unfortunately, the Virtual Cam does not have many settings to play with, but since OBS has more users than the Interceptor that should be something that might have been observed and maybe solved by other users...