Staacks/gbinterceptor #9: Tested Games

Megabusterretro, 2023-01-11:
Here is a list of the games I have currently tested and what games I tested it on.  Let me know if there is a better way for me to share this spreadsheet.  I will also upload video of each of the games with issues tomorrow and then post a link to my private youtube channel for you to view.

<html>
<body>
<google-sheets-html-origin><style type="text/css"></style>

Title | Model | Working? | CONSOLE TESTED | Notes
-- | -- | -- | -- | --
Bart Simpson Escape from Camp Deadly | DMG-TS-USA | YES | DMG SP | Works on SP not on DMG (Have VIdeo) SCREEN FREEZES AT TITLE SCREEN
Baseball | DMG-BS-USA | YES | DMG SP |  
BATMAN RETRUN OF THE JOKER | DMG-B5-USA | YES | DMG |  
BATTLEBULL | DMG-BR-USA | YES | DMG |  
BATTLETOADS AND DOUBLE DRAGON | DMG-VN-USA | NO | DMG | TITLE SCREEN WORKS, ONCE IN GAME SCREEN IS VERY DISTORTED
BUBBLE BOBBLE | DMG-B20USA | YES | DMG |  
BUGS BUNNY CRAZY CASTLE | DMG-BB-USA | YES | DMG |  
Darkwing Duck | DMG-DC-USA | YES | DMG |  
Defender/Joust (Arcade Classic) | DMG-ADJE-USA | NO | DMG SP | Image Ghosting on SP, DMG Unplayable
F1-RACE | DMG-F1-USA | YES | DMG |  
FIGHTING SIMULATOR | DMG-HR-USA | SORTA | DMG | SPRITE DEFORMED ON FIRST LEVEL, FIXES ITSELF AT BOSS LEVEL
FORTIFIED ZONE | DMG-IY-USA | YES | DMG | NEEDS FW 1.02
GARGOYLES QUEST | DMG-RA-USA | YES | DMG |  
HOME ALONE 2 | DMG-AH-USA | YES | DMG |  
JAMES BOND 007 | DMG-AWTE-USA | YES | DMG |  
KEN GRIFFY BASEBALL | DMG-AKGE-USA | YES | DMG |  
MARIO LAND 2 | DMG-MQ-USA | YES | DMG |  
Megaman III | DMG-W3-USA | YES | DMG |  
NBA ALL-STAR CHALLENGE | DMG-AC-USA | YES | DMG |  
NHL 95 | DMG-ANHE-USA | YES | DMG |  
OPERATION C | DMG-CN-USA | YES | DMG |  
POPEYE 2 | DMG-PG-USA | YES | DMG |  
PRIMAL RAGE | DMG-AR9E-USA | YES | DMG |  
Q-Billion | DMG-QB-USA-1 | UES | DMG SP |  
RAMPAGE (BLACK GBC CART) | DMG-ARPE-USA | YES | DMG | ONLY TESTED ON DMG GAMEBOY
REN AND STIMPY VEEDIOTS | DMG-VD-USA | YES | DMG |  
RESCUE OF PRINCESS BLOBETTE | DMG-RP-USA | NO | DMG |  
REVENGE OF GATOR | DMG-PB-USA | YES | DMG |  
ROCKMAN WORLD 4 | DMG-R4J | YES | DMG | JAPANESE VERSION
ROCKMAN WORLD 5 | DMG-WMJ | YES | DMG | JAPANESE VERSION
SAMURAI SHOWDOWN | DMG-X4-USA | SORTA | DMG | GAME IS PLAYABLE BUT HEALTH BARS ON TOP ARE CUT OFF
SHANGHAI | DMG-SH-USA | YES | DMG |  
Snow Brothers | DMG-SJ-USA | YES | DMG |  
STAR WARS (CAPCOM) | DMG-WS-USA | YES | DMG |  
TECMO BOWL | DMG-TL-USA-1 | YES | DMG |  
The Cool Spot Adventure | DMG-CU-U SA | SORTA | DMG SP | Ghosting while moving
TINY TOON ADV MONTANA'S MOVIE MADNESS | DMG-T8-USA | YES | DMG |  
WWF SUPERSTARS | DMG-WX-USA | YES | DMG |  
ZOOP | DMG-AZPE-USA | NO | DMG SP | GB-I CRASHES IMMEDATELY AFTER NINTENO LOGO WORKS ON SP

</body>
</html>

Staacks, 2023-01-11:
Wow, thanks a lot. Hopefully some of the crashes will be resolved with the fixes that I already created for the upcoming 1.0.3. I will go through these before release and then add them to the compatibility list.

About the ghosting: Is this the frame blending feature? If you press the mode button (next to the LEDs) twice (switches to green and back) it should be gone.
(I will add an on-screen display to make this button clearer and I will probably default to "off" as it seems to confuse more often than it helps, but it improves games that have fast flicker effects which do not translate well to the lower frame rate of the Interceptor video.)

Staacks, 2023-01-11:
No, if it is clear what happens I should be able to reproduce it with a flash cart. Also, I will probably get 1.0.3 ready this evening or tomorrow, which has a few fixes that hopefully also fix some of the problems you have found so far.

Staacks, 2023-01-11:
I just tried if some of the problems have been fixed with v1.0.3 and it seems like "Bart Simpson Escape from Camp Deadly" and "The Rescue of Princess Blobette" are fine now. Not sure about the latter, though. Can you confirm, because maybe I did not play far enough.

I still think that the two ghosting entries are the frame blending feature. The new version starts with it disabled and gives feedback on screen if you toggle it with the mode button.

About Zoop: Is GB-I the Game Boy Player? Does the problem only occur there?

Megabusterretro, 2023-01-12:
The GB-I is the GB Interceptor, sorry for the confusion!  

The OSD is great!  You are right, I was tweaking with the button on the back break up some of the filament because it was snug so I must have accidentally turned blending on.  Spot worked a lot better!

Updated the firmware, and Blobette is working.  Still having issues with Camp Deadly right at the beginning.  It seems a frame updates only when something makes the screen pause.  Here is the video link:

https://youtu.be/BTohWmj9F2M (apologies for the low volume).

Staacks, 2023-01-12:
> The GB-I is the GB Interceptor, sorry for the confusion!
Ah, lol, why didn't I figure it out myself...

> The OSD is great! You are right, I was tweaking with the button on the back break up some of the filament because it was snug so I must have accidentally turned blending on. Spot worked a lot better!
No, it was on by default, but you were not the only one who disliked it in some games, so I changed it to off by default in v1.0.3. Unfortunately, it is necessary in some games. If you look at this short recording of Zelda (https://twitter.com/diconx/status/1613308004136263681), have a close look at the dog chain. On the device it looks semi-transparent because each link of the chain is only drawn on every second frame. Without frame blending you only see every second link (they seem to alternate) because the 29fps of the Interceptor only show every other frame of the (almost) 60fps Game Boy. With frame blending it still is not perfect, but all links are visible at the same time.

> Updated the firmware, and Blobette is working. Still having issues with Camp Deadly right at the beginning. It seems a frame updates only when something makes the screen pause. Here is the video link:
> 
> https://youtu.be/BTohWmj9F2M (apologies for the low volume).
> 
> Zoop is no longer crashing but it has similar issues like the Simpsons. If I pause the game the screen updates, once I unpause it freezes again.

ok, I have to check both again. I am very sure that I did not see this issue, but I am not sure anymore if I tried it on the DMG or a GBC.

...and of course I have to check all the others, too. This issue will be open for a while :)

Megabusterretro, 2023-01-13:
For now, all my testing will be on my DMG so there is no confusion.  If you need me to test some specific games on other handhelds I will gladly do it.  I currently have a DMG, SP, and GBC.

Here is a few games I was able to test today.  I was on a PC without OBS so I decided to test the Win 10 Camera app..it worked like a champ to do some testing!  That is why you see another column annotating what application I tested with.

<html>
<body>
<google-sheets-html-origin><style type="text/css"></style>

Title | Model | Working? | CONSOLE TESTED | Notes | APPLICATION TESTED
-- | -- | -- | -- | -- | --
NBA JAM 99 | DMG-AJME-USA | YES | DMG | BLACK COLOR CART | WINDOWS 10 CAMERA
KIRBY'S DREAM LAND | DMG-KY-USA | YES | DMG |   | WINDOWS 10 CAMERA
SPOT | DMG-S3-USA | YES | DMG | CANT SEE GAME MODE SELECTOR BUT OTHERWISE GAME WORKS | WINDOWS 10 CAMERA
IN YOUR FACE | DMG-YF-USA | YES | DMG |   | WINDOWS 10 CAMERA
YOSHI | DMG-YO-USA | YES | DMG |   | WINDOWS 10 CAMERA
ROCKMAN WORLD 3 | DMG-W3J | YES | DMG | JAPANESE GB GAME | WINDOWS 10 CAMERA
NBA ALL STAR CHALLENGE 2 | DMG-H2-USA | YES | DMG |   | WINDOWS 10 CAMERA
STREET FIGHTER II | DMG-ASFE-USA | YES | DMG |   | WINDOWS 10 CAMERA
BLADES OF STEEL | DMG-UB-USA | YES | DMG |   | WINDOWS 10 CAMERA
MEGAMAN X TREME | DMG-BM6E-USA | YES | DMG |   | WINDOWS 10 CAMERA
DONKEY KONG LAND 2 | DMG-ADDE-USA | YES | DMG |   | OBS
MINER 2049 | DMG-MV-USA | YES | DMG |   | OBS
MOLE MANIA | DMG-MAMOE-USA | NO | DMG | SIMILAR ISSUES TO BART ESCAPE FROM CAMP DEADLY | WINDOWS 10 CAMERA
DEADALIAN OPUS | DMG-PU-USA | NO | DMG | SIMILAR ISSUES TO BART ESCAPE FROM CAMP DEADLY | WINDOWS 10 CAMERA
BASES LOADED | DMG-BK-USA | NO | DMG | SIMILAR ISSUES TO BART ESCAPE FROM CAMP DEADLY | WINDOWS 10 CAMERA

</body>
</html>

Staacks, 2023-01-13:
That the stock camera app works is good news indeed.

But now it gets weird: I just tried "Bart Simpson Escape from Camp Deadly", "Zoop", "Mole Mania", "Deadalian Opus" and "Bases Loaded" on my DMG and all of them work without problems (except for some garbled font in a menu somewhere). I am really confused now. I am testing on a simple flash cartridge but that should make no difference.

Looking at your video I am wondering if it really updates when you pause the game or if it is more a USB video glitch than an emulation glitch. After all, the frames that are shown look alright - they are just updated at an extremely slow rate. But in that case I would expect this to happen with all games, because the video format used by the Interceptor requires the same bandwidth for every image.

Is this reproducible for these particular games or could it be a random hickup (= still a bug) that just coincidentally occurred when you tried these games?

Megabusterretro, 2023-01-13:
I just swapped USB C cables and tested:
Escape from Camp Deadly (Works)
Zoop (works)
Bases Loaded (Works)

Well crap....I was using a Nintendo USB Cable...I figured that was the best one I had.  New cable seems to be working well.

I think I am going to widen the hold in the 3d printed cases to make room for some bigger sized USBC Cables.  I wonder if the cable wasn't fully inserted during some of these tests.

I am sorry for these false positives!

At least the Windows camera worked!

Staacks, 2023-01-13:
Damn, those cable dependent problems really worry me. I do not understand these at all.

Megabusterretro, 2023-01-13:
I agree, there are so many variables do doing this stuff.  I hope that doing this at least helps a little bit.  I definitely do not want to hinder the process you have going on here!

I still have more games to test, hopefully this revelation will yield some more positive results.  Going to stream/test games tonight with it!

Megabusterretro, 2023-01-13:
Here is a list of 22 games that been tested with the NEW USBC Cable!  Good results, just a few issues that I think you are already aware of on previous games.

<html>
<body>
<google-sheets-html-origin><style type="text/css"></style>

Title | Model | Working? | CONSOLE TESTED | Notes | APPLICATION TESTED
-- | -- | -- | -- | -- | --
Pokemon Pinball | dmg-bphe-usa-1 | YES | DMG |   | OBS
BONKS ADVENTURE | DMG-GK-USA | SORTA | DMG | SIMILAR ISSUE AS FIGHTING SIM.  MODEL IS SPLIT WHEN MOVING RIGHT | OBS
HEIANKYO ALIEN |   | YES | DNG |   | OBS
BATTLETOADS | DMG-BV-USA | NO | DMG | SIMILAR ISSUE TO BATTLETOADS/DD (VIDEO TAKEN) | OBS
BO JACKSON | DMG-BJ-USA | YES | DMG |   | OBS
A BUGS LIFE | DMG-APXE-USA | YES | DMG | BLACK COLOR CART | OBS
RADAR MISSION | DMG-RM-USA | YES | DMG |   | OBS
MALIBU BEACH VOLLEYBALL | DMG-SV-USA | YES | DMG |   | OBS
QIX | DMG-QX-USA | YES | DMG |   | OBS
ALIEN VS PREDATOR | DMG-A9-USA | NO | DMG | GAME CRASHES AFTER TITLE SCREEN VIDEO TAKEN (DMG) | OBS
MARBLE MADNESS | DMG-MB-USA | YES | DMG |   | OBS
SUPER MARIO LAND | DMG-ML-USA-1 | YES | DMG |   | OBS
MICKEYS DANGEROUS CHASE | DMG-MC-USA-1 | YES | DMG |   | OBS
THE SMURFS | DMG-UF-USA | YES | DMG |   | OBS
KLAX | DMG-KX-USA | YES | DMG |   | OBS
JUNGLE STRIKE | DMG-AJGE-USA | YES | DMG |   | OBS
METAL MASTERS | DMG-E6-USA | YES | DMG |   | OBS
BUBBLE GHOSE | DMG-BG-USA | YES | DMG |   | OBS
WWF SUPERSTARS 2 | DMG-WX-USA | YES | DMG |   | OBS
PAC-MAN | DMG-PC-USA | YES | DMG |   | OBS
TETRIS 2 | DMG-EH-USA-1 | NO | DMG | LESS THAN HALF THE SCREEN IS DISPLAYING (VIDEO TAKEN) | OBS

</body>
</html>

Staacks, 2023-01-14:
> I agree, there are so many variables do doing this stuff. I hope that doing this at least helps a little bit. I definitely do not want to hinder the process you have going on here!

Don't worry, it really helps a lot. Even if I went through all those games myself I would miss system specific stuff like the cable issue or Windows-related things.

About the cable problem: Could you test this firmware with the old cable? https://cloud.tankwar.de/index.php/s/8qEMYjzMcHJEibx

This firmware is meant to fix issues with boards where the oscillator takes a bit too long to settle after turning on. This affects individual board because of manufacturing tolerances for some components and the symptoms are that the Interceptor would work after uploading a new firmware, but stops working once it has been unplugged and connected again (that's when the original firmware would not wait long enough).

I am now wondering if there might even be an edge case where the oscillator has just settled enough to be able to start, but the PLL or VCO would not be set up correctly for stable USB communication. I have to admit that my knowledge here is very limited (and the little knowledge I have says that this is not possible) and this is a rather long shot, but different combinations of cables and games (which have different PCBs and chips) might influence the capacity enough to tip it over the edge.

Thanks for all the new tested games. I will update the list and eventually start looking into which can be fixed easily before getting into game specific fixes...

catrenelle, 2023-01-15:
@Megabusterretro Want to combine efforts? https://docs.google.com/spreadsheets/d/1t-xKYgYn_4Pzesktp5r916mh_z9X_GMTLp_1bZnPkp0/edit?usp=sharing

Megabusterretro, 2023-01-15:
> @Megabusterretro Want to combine efforts? https://docs.google.com/spreadsheets/d/1t-xKYgYn_4Pzesktp5r916mh_z9X_GMTLp_1bZnPkp0/edit?usp=sharing

Sure thing!  I will migrate things over to that list!

Megabusterretro, 2023-01-15:
> Don't worry, it really helps a lot. Even if I went through all those games myself I would miss system specific stuff like the cable issue or Windows-related things.
>
> About the cable problem: Could [...]

So I tired The Simpsons Escape from Camp Deadly with the "bad" USB C cable and it worked with 1.0.4-alpha1.

Staacks, 2023-01-15:
@catrenelle Am I correctly thinking that you are "CatHouse Games" from Twitter?

I mention and thank you, "Joe Ray Six" and @Megabusterretro on the [game compatibility list](https://github.com/Staacks/gbinterceptor/wiki/Game-compatibility). Let me know if you prefer a different name or a different link target.

> Sure thing! I will migrate things over to that list!

ok, then I will merge your last list to the Wiki and then work from Google Docs to add new titles and see what still needs to be fixed.

> I tired The Simpsons Escape from Camp Deadly with the "bad" USB C cable and it worked with 1.0.4-alpha1.

That's good to hear. I am not sure if these cable issues are 100% reproducible, but it makes some sense to me. I guess I will only know if I see no more issues that are resolved by changing the cable in the future.

catrenelle, 2023-01-15:
Correct! "Joe Ray Six" is a friend of mine as well and has been helping to test a lot. I'm trying to get fabrication costs of the unit down as much as I can so I can make and sell them for cheap in small batch quantities. But we're also testing the games and welcome anyone who wants to contribute to the game list. Just submit an editor request and we'll review/grant as appropriate.

Staacks, 2023-01-15:
Just a little info as my current findings do not result in new fixes at the moment. It seems like I found most generic bugs by now and start to stumble upon problems that need to be fixed for individual games. For example:

Alien vs. Predator does not return from a wait function after a DMA transfer, but keeps doing a few things while executing from RAM. The problem is not that the Interceptor does not know which commands are in RAM, but that it cannot see their execution during DMA and cannot know which of the addresses is a command and which just data. I cannot think of a generic fix for this, but a game-specific one would be super easy as the Interceptor can just look for a specific hardcoded address to be read.

Bonk's Adventure is one of those cases where data from an I/O register is directly used for graphics. The input from the gamepad is bitmasked and shifted and becomes the bit that flips sprites when you turn in a different direction (probably the same for fighting sim). Definitely no generic fix here, but since changing directions also changes the tiles which seems to work on the Interceptor, a game-specific fix seems possible.

So, I think that at this point I will stop looking into glitches for a moment and look into detecting which game is running. This is a bit tricky, but individual fixes are only possible if the Interceptor knows the current game.

Megabusterretro, 2023-01-16:
@Staacks I am glad you pointed out the Bonk's issue!  The same thing is happening on Fighting Simulator.  The sprite gets very strange when moving left to right, but looks normal when moving right to left.

Also, I found a pretty odd bug with Ultima: Runes of Virtue.  When booting the game up with the Interceptor I get a black screen and a loud buzzing noise.  This happend with multiple interceptors on multiple versions of the firmware.  I also tested this on an SP and the DMG.  I am guessing it is late in your neck of the woods so I will post a video in a little bit.

Staacks, 2023-01-16:
I think at the moment while there still is so much to do, the list works better.

My mail notification shows the unedited version of your comment: Has this been resolved? Sounds worrying if it's not just a loose contact...

Megabusterretro, 2023-01-16:
> I think at the moment while there still is so much to do, the list works better.
> 
> My mail notification shows the unedited version of your comment: Has this been resolved? Sounds worrying if it's not just a loose contact...

It is for sure still an issue but I didn't want to bring it up until I had the proper method on how you wanted it reported.  I took some video of the issue and I can post a youtube link in the spreadsheet.

Staacks, 2023-01-16:
Yeah, a video might be helpful, but also any thoughts on what might be different here. If I understood correctly it is an issue on the actual Game Boy and not the output from the Interceptor? Since the cartridge has a direct connection to the Game Boy through the Interceptor and the bus transceivers are hardwired to only output in direction of the rp2040 that should not happen. So, I am a bit of a loss here. I also do not expect to be able to reproduce it with a flash cartridge, but I will try later...

catrenelle, 2023-01-16:
> 

You should be able to request editor access to the list when you have it open.

Staacks, 2023-01-16:
> I do not have access to update that list yet, so here is the video of the Ultima issue.
> 
> https://youtu.be/nrQJmS8FMZA

Please tell me that this sound is coming from the speaker (i.e. you can control it with the volume dial). It is terrifying...

As expected, I am not able to reproduce it with a flash cart, but since it happens even before the Nintendo logo, I did not expect that this has anything to do with the ROM code. This is some kind of hardware issue.

One interesting thing to note is that the Interceptor does not even detect that the Game Boy is turned on. Not sure what that means, but since I am at a total loss I am just collecting information at this point.

A few things to try to maybe get more infos to narrow it down a bit:
* Does this also happen when the Interceptor is not plugged in? (I mean that the Interceptor is between the game and the DMC, but no USB cable plugged in.)
* Do you have a copy of "Top ranking tennis" that you could test? This is a bit of a long shot, but I am wondering if maybe the power on detection of the Interceptor (which effectively is just a pull-down resistor) might be interfering with the supervisor chip on the cartridge. I found an [image of the PCB](https://www.pricecharting.com/de/game/gameboy/ultima-runes-of-virtue-ii) and it seems to be a BA6129 while most other games have a BA6129A or entirely different chip. I could only find "Top ranking tennis" in [Gekkio's hardware database](https://gbhwdb.gekkio.fi/cartridges/) with a BA6129 and some versions of Zelda and Mario 2, but those come in many different revisions. Unfortunately, Ultima and this tennis game might also come in many other revisions, so even if you have it, this test is only a hint at best.

Staacks, 2023-01-16:
Oh, a few more thoughts and tests about the BA6129 hypothesis:
* Do the  batteries in the DMG make a difference? (fresh vs. older ones)
* Do you know the state of the cartridge battery? (i.e. have you ever replaced it?)

Megabusterretro, 2023-01-17:
- Sound was coming from the Game Boy speakers...yeah it was pretty jarring the first time it happened.
- This is only happening while the Interceptor is plugged in via USB-C
- I do not have Top Rank Tennis
- Battery is still working, I was able to load the saved game.
- I just loaded some brand new batteries on the game boy and oddly enough, I have not had the error.  I was using a combo of rechargeable batteries and normal batteries prior.  

This is such crazy randomness....I never would have thought batteries could cause an issue like this.

Megabusterretro, 2023-01-17:
I lied, I do have Top Rank Tennis and it is doing the **exact** same thing Ultima was doing.  I put the new batteries in and it fires up just fine!
[image]

Staacks, 2023-01-17:
Perfect. It begins to make a bit of sense. So, this seems indeed related to the BA6129. Unfortunately, I could only find datasheets for newer models, but I think that your batteries are just at the threshold of the BA6129. A little lower and it would keep the reset line for the Game Boy active, not allowing it to turn on anymore, and a bit higher and it would properly switch on. Somehow it is oscillating between both states and I am wondering if it might be ripple from AC that is pushing it over the edge (although the noise is not 50Hz or 60Hz for the US, but something around 79Hz).

In the end, it will probably be filed away as "cannot be helped" (at least not without a hardware revision that adds a filter), but could you try the following just for a more complete picture:
* Can you measure the voltage of the old batteries?
* Can you try (with the old batteries) to use the Interceptor with a laptop (or Andriod phone with USB-C) that runs on its internal battery (not connected to AC at all).
* Can you try (with the old batteries) if the problem occurs when you do not plug in the Interceptor, but instead just ground the USB cable. (i.e. one end is plugged into the Interceptor and you hold the outside of the other plug to the metallic backplate of a PC - could be a bit trickier if you have a laptop with plastic case).

Megabusterretro, 2023-01-18:
- I have misplaced my meter!  I will test the batteries once I find it.
- I did that and got the same buzzing sound.  I did notice that it worked with the old batteries right as I inserted them.  Both games fired up with no issue.  I went to look for my meter and check on my 3D prints, turned it on again and got the buzzing noise.
- Tried grounding the cable on metal and the game fired up with the interceptor attached to the USB-C cable.

Staacks, 2023-01-18:
Hm... Not sure what to make of this. I expected that ground plays a role here. Anyhow, at least it seems to be narrowed down enough to not be a major problem. I will add it to the trouble shooting guide.

JJ-TerhoA, 2023-01-22:
Title || Model || Working? || Console tested || Notes || Application tested

Dr. Mario || DMG-VU-ESP || Almost working || DMG || Objects on playfield at the start of game not showing correctly. Majority of objects appear after first 4-line cleared, but not necessarily all || OBS and Kamera

SolarStriker || DMG-SS-SCN || Yes || DMG || - || Kamera

Chase H.Q || DMG-HQ-SCN || Sort of || DMG || Road and grass not uniform as should be, but otherwise game is playable || Kamera

Tom & Jerry, Frantic Antics! || DMG-J8-USA || Yes || DMG || - || Kamera

Staacks, 2023-01-26:
Damn, fixing the remaining games get's more and more tedious.

The bad news: I spend several evenings thinking about the Dr. Mario problem and I think it is the first game I give up on - at least for now. Unless I figure out an exact emulation for the div register, it seems to be unsolvable with the available resources. The problem is that it generates the level and writes plenty of random (= div register) data directly to VRAM and RAM without conditional branching.

The good news: I fixed Bonk's Adventure (see https://mastodon.social/@DiConX/109757439865554681 and https://mastodon.social/@DiConX/109757442170688549). This one creatively rearranges the bits of the gamepad input to set the mirror flag of the player tiles. So, I implemented an option to set an address specific to a game that the Interceptor tries to reconstruct by observing conditional jumps that are based on this address.

Staacks, 2023-01-28:
A few more fixes will be in 1.1.0:
- Defender (Arcade Classics 4)
- Chase H.Q.
- Battletoads (all three of them)
- Kirby's Dream Land 2
- Teenage Mutant Ninja Turtles

I think I will soon be through with the games posted here and then move on to the Google Docs list. (Well, soon depends on the games. Battletoads cost me an entire evening digging through the game to eventually find a bug in my implementation that is fixed by replacing a single character in the code...)

Megabusterretro, 2023-01-29:
That's great news!  I have been building my own list of games I've tested and will add to the sheet soon. I am also getting some Japanese games in next week I look forward to testing!

Megabusterretro, 2023-02-01:
Updated the spreadsheet with my game collection (Minus the ones listed in this thread).  Is there a Discord server to collaborate on or is Github the main source?

Staacks, 2023-02-03:
Not sure how you testers want to work. I eventually place everything in the Wiki here, because that's where others will look for information, but my list is a bit simpler and does not track testing progress across different systems.

I have now merged everything from Google Docs to this list and am working on the remaining issues. Some games have been fixed with other fixes and I aim to fix all known issues except for really difficult ones (Dr. Mario) for the release of version 1.1.0.

I had difficulties to reproduces two problems, though:
Balloon Kid: Player sprites look fine on GBC. Does this occur in a specific situation or part of the level?
Bust-A-Move 2: The bonus scores looked fine to me. I think I have seen five of them. Is it particularly rare?

Scrapz, 2023-03-09:
I figured I'll add my issue here as it seems relevant? Feel free to tell me otherwise.

I'm unable to get "Yoshi" working properly, in particular the PAL version which is called "Mario & Yoshi". Basically the same thing, except the model is "DMG-YO-AUS" as opposed to "DMG-YO-USA".

It works normally in the Gameboy, but when going through the inteceptor I get a slightly garbled Nintendo logo on boot, and the game doesn't load, and it only does it to this one game. I can play Tetris no problem, I can play Donkey Kong Land no problem. Just for some reason it corrupts Mario & Yoshi. My other games are also PAL (ie; DMG-YT-AUS) so I don't think it's a PAL compatability thing.

Is there anyone else out there with the PAL version of Yoshi that can confirm?

Staacks, 2023-03-09:
If the Nintendo logo is scrambled it's not a software issue of the Interceptor - the Interceptor does not do anything at this point and the pins of the cartridge are supposed to be directly wired to the Game Boy anyways. If the logo is scrambled the Game Boy is not able to properly read the cartridge.

If it only happens in the combination of the Interceptor and that particular game I suspect that it is pretty much the same as any scrambled Nintendo logo, but in an unlucky pattern that it only shows with the Interceptor. Think of the typical dirty (or maybe even corroded) contacts of a cartridge and a slightly different alignment/tolerance of the contacts on the Interceptor. The Game Boy may have proper contact as it touches a clean part of a pin while the connector of the Interceptor touches the same pin a bit further down or slightly to the side or with slightly less pressure - and you have a game that works with the Game Boy but not with the Interceptor. If other games work with the Interceptor I am quite sure that it is the connection between the game and the Interceptor. So, try to clean the contacts (if you have not yet done so) and check if the game can be wiggled around a bit inside the slot (careful, if you do so with bad connections while the Game Boy is turned on - that could corrupt your save games on that cartridge).

MelonGx, 2023-06-02:
Hello I have some tested game glitch or don't work.

Pokémon Silver: Black & White mode on GBC GBA GBASP.
Pokémon Pinball: Black & White mode & glitch on GBC GBA GBASP.
Everdrive GB X7, Tokimeki Memorial Pocket Culture/Sport versions, Heroes Might & Magic II: Error.

Staacks, 2023-06-02:
Thanks for reporting issues. However, those are all Game Boy Color games which are not expected to work with the GB Interceptor. Even if some of them are hybrids that work on a classical Game Boy, they will only run in classical mode if they are played on a classical Game Boy (DMG, Pocket or Light) and are still expected to crash on a GBC. Setting a black and white palette on the GBC also does not make a difference here.

But I am wondering if I am missing something from your report as you say "Black & White mode"... Is there a trick to set a color palette on color games like when playing classical games on GBC? Even then I doubt that the games are running in any different way (in particular in double speed mode which the Interceptor cannot handle).

The Everdrive GB X7 is an interesting detail here. I suppose it also has a color mode for the menus? Does it start working when you select a classical game from the menu? i.e. the Interceptor shows an error while you select a ROM on the GBC screen and when the Everdrive starts the ROM (a classical one!), the Interceptor starts working?

MelonGx, 2023-06-05:
> But I am wondering if I am missing something from your report as you say "Black & White mode"... Is there a trick to set a color palette on color games like when playing classical games on GBC? Even [...]

They are all forced output in DMG mode, even in GBC GBA GBASP.

Staacks, 2023-06-05:
How are they forced to run in DMG mode?

MelonGx, 2023-06-05:
>How are they forced to run in DMG mode?
By default.

Staacks, 2023-06-05:
Sorry, I do not understand what you are doing. You put a regular Pokemon Silver cartridge into a regular GBC and it does not show any colors? That does not seem like "default" behavior to me.

MelonGx, 2023-06-05:
> Sorry, I do not understand what you are doing. You put a regular Pokemon Silver cartridge into a regular GBC and it does not show any colors? That does not seem like "default" behavior to me.

But I do receive this result. I can't find any instruction to change it.

Staacks, 2023-06-05:
And this happens without the Interceptor with Pokemon Silver and Pokemon Pinball on GBC, GBA and GBA SP?

MelonGx, 2023-06-05:
> And this happens without the Interceptor with Pokemon Silver and Pokemon Pinball on GBC, GBA and GBA SP?

All my GBC compatible & GBC only softwares are either forced running in DMG mode, or crash then error.

Staacks, 2023-06-05:
Wait, are you talking about the output from the Interceptor? But the GBC screen is showing colors, isn't it?

MelonGx, 2023-06-05:
> Wait, are you talking about the output from the Interceptor? But the GBC screen is showing colors, isn't it?

Of course it is.

MelonGx, 2023-06-05:
I understand it. Thank you for letting me understand how the interceptor work.

Widge-5, 2023-07-21:
Couldn't see **Pocket Bomberman** on any list of tested games, so I can confirm that it does work apparently flawlessly with the GB Interceptor.
It's a hybrid game so won't work when used with a GB Color, but perfect on a DMG.  Tested with OBS.
I think I have a few other games not listed in the compatibility list so I'll check those out soon.

Staacks, 2023-07-22:
Thanks, I have added it to the list.

Widge-5, 2023-07-23:
I've been through my collection and tested the following games not already named on the compatibility list, using a DMG and the GB-Interceptor displaying in OBS.  Firmware version 1.1.0.
All games are original GameBoy games (no GBC-hybrids) and they all seem to display correctly through the Interceptor.  

- The Simpsons - Bart vs. the Juggernauts
- Bionic Commando
- Desert Strike - Return to the Gulf
- Game Boy Gallery - 5 Games in 1
- Mario & Yoshi
- The Amazing Spider-Man
- Spider-Man 2 (aka The Amazing Spider-Man 2)
- Spider-Man 3  (aka The Amazing Spider-Man 3 - Invasion of the Spider Slayers)
- Star Wars (not Capcom) - _the compatibility list explicitly refers to a "Capcom" version. AFAIK, only one game called Star Wars was released for the original GameBoy but mine makes no mention of Capcom. I believe this is the same game._
- Wave Race

Staacks, 2023-07-25:
Thanks, I have added them to the list.
I also removed the "Capcom" label in the list. I am no expert in different version of the same game, but it seems like there was a DMG-WS-USA and a DMG-WS-USA-1, which seem to be the same game, but one was published by Capcom and one directly by Nintendo (?). Anyhow, I am not keeping track of different versions in that list, especially since I am not aware of a case where one version of the same game works and another doesn't - so it's only "Star Wars" now.

TorqueEmPup, 2024-11-06:
TMNT Fall of the Foot Clan is not without issue like stated in the compatibility chart.
The text in the intro, and the scenes between levels is flickery and just downright broken.

[image]

Different USB-C cables don't affect it at all.
I don't have an original cart, this is off an Everdrive X3