Staacks/gbinterceptor #11: Feedback on PCB design

remy, 2023-02-01:
**Describe the bug**

Not a bug, more of an "in the field" use report.

One of the boards I had made I sent to a youtuber (RetroBreak) just as a thing to show off, but the first one didn't work. Couldn't work out why, but when he returned it (I sent him a new one) it turned out the pads had ripped up on the connector.

It's carnage! 

[image]

On inspection I can see how it happened. The cart port was not 100% flat - I think there's about 0.2mm tilt. I'm not sure if that was from cart insertion or from not being entirely flat when the port was soldered down (I suspect the later).

I wondered if vias under the cart port pins would be more secure (under the actual cart port rather than below to reduce stress if it's bent) - at the same time just "be more careful" is a simpler/cheaper approach.

Obviously feel free to close this right away, just reporting it for you.

Staacks, 2023-02-01:
Well, it sounds like something I should warn about in the build guide, but I am not entirely sure yet if I understand correctly how that happened.

I assume that you have soldered the two large pads on the other side and that it is one of the cartridge slots with the two pins that fit into the holes in the PCB? I mean, I currently spent most of my evenings pulling out and re-inserting my flash cartridge to test and fix various games and my version still seems perfectly robust. Not to speak about the absurd contraption of testing a Game Genie with the Interceptor.

Was this in a 3d printed case, so that stress might be added from an edge of case pressing on the cartridge?

remy, 2023-02-01:
You know what, I now think it actually got crushed during transit when I posted the cart to the guy.

I was sure it was working before I sent it, and now looking at the cart slot side by side with another (working) Interceptor, I can see it's actually bent a little:

[image]

Hmm. Random bad luck on my part.

So completely ignore me - the fact that you're inadvertently stress testing (to me) definitely means there's no design flaw. 

I guess the warning now is: _do not sit on GB Interceptor - does not work as a chair_!

closed 2023-02-01 by remy

Staacks, 2023-02-01:
:laughing: