Staacks/gbinterceptor #16: New PCB Assembly, DMG Is in a Boot Loop

Megabusterretro, 2023-02-28:
**Describe the bug**
I just purchased another batch of Interceptors.  The first two I assembled are caught in a reboot loop when I try to use them.  I have tried 3 different DMG Consoles, however, it works with an SP and several Game Boy colors.

I am including pictures of the board.
**To Reproduce**
Steps to reproduce the behavior:
1.Insert Interceptor and Game Cart to DMG Game Boy and power on

**Expected behavior**
Expectation is for the DMG to Boot

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]
[image]
[image]

Megabusterretro, 2023-02-28:
Just an update.  I did a third one with the same results.  DMG is in a constant reboot loop.  Here is a better view of the board as well.
[image]
[image]

Megabusterretro, 2023-02-28:
https://user-images.githubusercontent.com/122132860/221760840-fdb2e3f1-45ad-4eb3-8b4a-fc688e21fc28.MOV

Staacks, 2023-02-28:
ok, that's odd, especially if it is working in a GBC and SP. No idea right now, but let's clarify a few details so I am thinking in the right direction:
- Working on GBC and SP means that the "intercepted" game also shows up correctly on a PC?
- Does it also loop when no game is inserted?

Megabusterretro, 2023-02-28:
- It is showing up correctly on the PC when booting with a GBC
- No loop when there is not a game inserted.

Staacks, 2023-02-28:
Yeah, I am looking for a defect or error that would significantly increase the current draw. The GBC can handle higher currents much better than the DMG (or the GBP which is the worst in that regard) and I assume that the SP should be be similar to the GBC in that regard.

Do you have a multimeter with which you can measure the resistance of R8 and R9? Those are a voltage divider to split the 5V from the cartridge to 3.3V so the rp2040 can detect if the Game Boy is turned on. They should be 5.5kOhm in total, so there is not much current flowing that would burden the Game Boy. But if for some reason they picked the wrong resistors here, that current might be much higher - too much for the DMG, but managable for the GBC.

So, if you measure the resistance across the outer ends of R8 and R9 together you should see 5.5kOhm (2.2kOhm just for R8 and 3.3kOhm for R9). If it is significantly lower that might be the problem.

Well, it is a rather wild hypothesis...

Megabusterretro, 2023-02-28:
Just rung the two out and they are reading 3.3 and 2.2 respectively unfortunately.

Megabusterretro, 2023-03-01:
I submitted a ticket with JCLPCB last night as well so hopefully they can figure out something

Megabusterretro, 2023-03-02:
Another update!  Looks like this is not working on a Game Boy Pocket either.

Staacks, 2023-03-03:
> Another update! Looks like this is not working on a Game Boy Pocket either.

That does not really surprise me. If whatever causes the problem triggers a reboot because it drains too much power, then the GBP is the first one to go, because in my experience with the Wifi cartridge it is the worst in handling power spikes.

Megabusterretro, 2023-03-03:
So I am working with JCLPCB but in the meantime I am going to double check my work and make sure it isnt just a bad soldering job on my part.  If it were something with how I was soldering the pins, which pins should I focus my attention on?

Staacks, 2023-03-03:
I don't think that a soldering error would be selective to the DMG and GBP. If data pins or address pins were affected you would not see the Nintendo logo (except maybe for some of the higher address pins, but those would cause problems on the GBC eventually). You also would not see the Nintendo logo if 5V, CLK or GND were not connected and other non-connected pins would either mean it always works (maybe no savegames with nWR) or the Nintendo logo would not work.

Shorted pins might be more interesting as some rare combinations of pin uses could lead to a short then, but I do not see any combination that would not also cause trouble for the logo or the GBC. The only interesting one might be the nRST pin, but again: Why would that not be a problem for the GBC.

I am thinking that turning on the ble LED somehow might trigger the problem. That would at least explain why it happens in that exact moment, but then again there are several things happening in that moment. Does anything happen if you press the mode button while the Nintendo logo ist still crawling down? That button also connects the LED to ground turning it on - that's not the same as turning on the LED with the rp2040, but maybe something happens in that case...?

Megabusterretro, 2023-03-06:
@Staacks Mystery Solved.  

I am a novice at soldering and I decided I wanted to use flux to give me less change of bridging contacts while soldering the 32 pin connector.  I guess on my first three from my second batch didnt burn off all of the flux I used and it was causing a connection issue.  I have done 6 more without flux and they work like a charm.  I also went back to a couple that were only working on the GBC/GBA and reflowed all the contacts and made sure I burnt off the rest of the flux and they started working.

Thanks for all of your responses....this has been a huge learning experience for me.

closed 2023-03-06 by Megabusterretro