Staacks/gbinterceptor #20: gb interceptor works in pc but it doesn't recognize the game

ScorpioNMsX, 2023-07-16:
**Type of issue**
gb interceptor works in pc but it doesn't recognize the game 
on the screen of the gameboy you can see the incomplete logo, but when you disconnect from the pc it does start correctly
**Describe the bug**
gb interceptor works in pc but it doesn't recognize the game 
on the screen of the gameboy you can see the incomplete logo, but when you disconnect from the pc it does start correctly

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error in logo

**Host software (please complete the following information):**
 - OS: Windows
 - Software: OBS

**Game Boy:**
- Model: gameboy color cgb 001

**Game:**
- Title: [e.g. Tetris]
- Version: [Usually printed on the cartridge, e.g. DMG-ZL-NOE]

**Additional context**
I have assembled all the components myself, could it be an assembly error?

Staacks, 2023-07-17:
To be sure I understood the details correctly: The Nintendo logo on the Game Boy is broken when the Interceptor is used and plugged into a PC. But if you just unplug the USB cable from the PC but the Interceptor is still between the Game Boy and the cartridge, the logo looks fine and the game starts?

That's an odd one, because the Interceptor is designed to not influence the game at all and the contacts of the cartridge should be wired directly to Game Boy. So, normally, if the logo does not work, I would indeed suspect a lose soldering point or a short between two soldering point. However, the state of the USB cable playing a role makes me a bit less confident about that

When you say that you assembled all components yourself, I assume you really mean ALL components, including the rp2040, USB socket, capacitors etc..? Does the Interceptor work on its own? So, can you see the standby screen?

ScorpioNMsX, 2023-07-17:
If correct, the gb interceptor without connecting to the pc without receiving power and inserted in the gameboy works but once connected from the obs the pc detects it and the image of insert game appears. when turning on the gameboy the nintendo logo is corrupted, the game does not start and the obs screen stays waiting for the game.
PS: I have assembled all the components myself including the rp2040 with soldering station. Could it be that some component is not well soldered? but in this case the pc should not detect the gb interceptor?

Staacks, 2023-07-18:
There are many ways thinks can break if there is a bad connection or a sort circuit. I would agree that most would end with the Interceptor not being recognized by the PC at all, but in some cases weird things are possible. For example, if there is a short between an actively driven GPIO output (for example the one that drives the blue LED, although that becomes active after the game has been recognized) and one of the cartridge pins, I would expect the game to fail when that LED turns on, which would only happen when the USB cable is plugged in.

So, that could be a tedious search... Do you have a voltmeter? Could you connect the Interceptor via USB and check if there is a voltage across any of the cartridge pins when it is not plugged into a Game Boy. (Actually, I am not 100% sure if a harmless potential difference is to be expected if the pins are not driven and since I am on vacation I do not have a working Interceptor at hand to compare. Still, it might be interesting if only one pin has a voltage against ground and others don't.)

Oh, and is the broken logo always the same? If so, could you post a photo of it? Maybe this helps figuring a specific pin that is always high or always low.

Staacks, 2023-08-05:
Any news on this? Since I suspect that it is a problem specific to your individual build I would close this for now, but let me know if I can be of any further help and please report back if you find something that might help others.

closed 2023-08-05 by Staacks