Staacks/gbinterceptor #21: Feature request - Additional colour palettes.

Widge-5, 2023-07-21:
Hopefully a relatively straightforward feature request (depending on how the colour palette is managed).  Could it be possibly to add alternate colour palettes into the firmware?
As the button on the Interceptor will currently toggle between B&W and Green images, I suggest adding more options for variety of displaying the 4-colour palette of the DMG GameBoy.
I imagine this as being similar to the GameBoy Color's palette selection for classic monochrome games.
I hope that makes sense.

Staacks, 2023-07-22:
Unfortunately, it would not work as flexible as what you are probably hoping for.

Internally the GB Interceptor has to manage three buffers with images: The one it is currently rendering to, the one that has been finishes that is ready to be sent via USB immediately when the next frame is due and the frame that is currently being sent via USB. (The last two cannot be the same because the USB video frame rate is not synchronized to the Game Boy frame rate).

In the upcoming version 1.2.0 with MJPEG support it is actually four buffers, because there is an additional one: One buffer that is currently rendered, one finished image that is currently encoded to JPEG, a JPEG version that is ready to be sent and the buffer that is currently being sent. In this case the latter two are a bit smaller because they are already in JPEG format, but in total it amounts to roughly the same amount of data.

The point in both cases is that these buffers take a lot of RAM and therefore only hold a grayscale component. Also for each processing step the Interceptor only needs to handle a grayscale component. The old NV12 format as well as the new MJPEG format both encode luminosity (grayscale brightness) and chroma (the color information) separately, so the Interceptor only handles luminosity and adds static color information in the very last step. It stores the color information only in the last two buffers and uses the same entry for every single pixel - only changing it once you switch from grayscale to green mode.

All this means is that assigning a specific "hue" to all grayscale pixel (i.e. making them all different shades of the same color) is very easy. Assigning differen colors (like light gray is yellow and dark gray is red) would be extremely difficult or maybe impossible because every buffer would need color info and every processing step needs to handle more data. Maybe we could leave the first buffers grayscale and assign the colors later, but that would mean that the data is no longer written sequentially and cannot be simply handled by a few DMA channels as it is right now.

Oh, also currently the color channels are subsampled by a factor of 2 (again, both in NV12 and MJPEG). This means that a block of four pixels will have to have the same hue anyway.

So: Are you looking for other single-colored options (like red mode in addition to green mode) or a palette option like the GBC provides?

Widge-5, 2023-07-23:
Thank you for the detailed explanation. It does help to understand how the Interceptor works.
My hope was that the Interceptor could give palette options in the same way as the GBC, but it appears that this isn't possible.
If additional single-colour options are added in to a future update then I would welcome that as a bonus. But overall I am very pleased with the Interceptor as it is with it's b&w and green modes.
Thank you for sharing your amazing project with the rest of the world :)

Staacks, 2023-07-25:
Glad you like the project :)
I don't think that I will add more color options, though (would love to add palettes, but that is tricky as mentioned above). Since the Interceptor only has a single button, toggling through multiple options would be a bit annoying when most people are only interested in gray and green. However, this can easily be achieved in OBS with filters.

closed 2023-07-25 by Staacks