Staacks/gbinterceptor #22: showed waiting the game.....

hpx1984, 2023-09-02:
Hi Sir

many thanks your hard working on this great project 

now i built one and every thing seems good 
but i just can saw the screen of "waiting the game....." or "game boy turned off"

where i can check to try to fix this issue?

hpx1984, 2023-09-03:
more information

i can saw the device on PC
[image]
[image]

FW:1.2.0
Turned off GameBoy
[image]

Turned on Game Boy ,the game boy working as normal but PC screen showed "waiting for game"
[image]

Interceptor status:
PWR Blue LED on,and another red LED is flash 
[image]

[image]
[image]

Sir ,any time ,please help me if you got my message and if you have free time

hpx1984, 2023-09-03:
hi Sir

i found the problem is on the flash IC , now fixed the problem and it can working 
[image]

closed 2023-09-03 by hpx1984

Staacks, 2023-09-04:
Glad you found the problem :)