Staacks/gbinterceptor #28: Windows Camera photo mode doesn't work but video does

Butterscotch96, 2024-05-06:
**Describe the bug**
Not sure if host software issues count as bugs but it's worth a shot. I want to use the Interceptor to easily get photos off my Gameboy Camera and I thought a good way to do that is to use the Windows Camera app. Videos work just fine but photo is just a blank screen.

**To Reproduce**
Steps to reproduce the behavior:
1. Use Interceptor like normal
2. Open Windows Camera
3. Select photo
4. See blank screen

**Expected behavior**
It should show up like normal but obviously a blank screen isn't normal.

**Screenshots**
[image]
[image]

**Host software (please complete the following information):**
 - OS: Windows 10
 - Software: Windows Camera

**Game Boy:**
- Model: AGS-001

**Game:**
- Title: Gameboy Camera
- Version: Can't find it sorry TwT

**Additional context**
I'm a nerd

Staacks, 2024-05-07:
Without any testing I would suspect that this is because the GB Interceptor does not implement the still image capture feature of the USB Video Class. Since the MJPEG stream is already lossless and since it already represents the full resolution of the Game Boy, still image capture is not really needed. For normal cameras it is used to obtain a single image with a higher resolution / better quality than the video stream, but for the GB Interceptor any frame from the video stream already has the maximum resolution and quality.

However, I will leave this open as a) I am not 100% sure that this is the reason for the problem (I would have expected that the preview would at least show up) and b) I can see how having a still image option might be convenient in some cases. Unfortunately, I do not see that I will find the time to look into this any time soon.

Butterscotch96, 2024-05-13:
What I did for now is I did a short video and then just used a random website to get all the frames as PNGs. Takes a while but works fine.

Unessential, 2024-06-14:
This might be related:

Even though all my laptop ports are supposedly USB 3.2 I found that one port had excessive audio delay when on vlc and obs couldn't capture at all.  (i'm on linux though).  switching to another  port worked for me. I suspect it's a power issue. since it should work with usb 2.0 right?

are you using any usb hubs? Are you on a desktop? or laptop?