Staacks/gbinterceptor #33: Question: would be possible to intercept high ram writes?

odelot, 2025-04-07:
Hi! I am the developer of the NES RetroAchievements Adapter. It brings internet and achievements to the real hardware https://github.com/odelot/nes-ra-adapter. 

Your way to intercept the BUS was an inspiration and indicate that I probably was on the right track when I tried to intercept memory writes with the PIO. Great work you did. Thank you for sharing all of this.

Which bring us to the Gameboy. I made a quick PoC to test the approach on gameboy and it worked very well (https://youtu.be/6lw4mGO8Hv80) to intercepts memory writes on the working RAM. 

The problem is the RA developers uses a lot of memory information from high ram to design their achievements. And this memory is not exposed on the cartridge slot / external bus.

The question is: do you think it would be possible to intercept the high ram writes? It seems you are intercepting the op codes from the game rom, is that right? Would be possible to use the same approach for storages on the high ram?

Thanks in advance.

Staacks, 2025-04-07:
Oh, that's a very cool project!

So, your problem is that you do not do the entire process of observing CPU instruction like the Interceptor does and therefore you need to observe RAM writes directly? Immediately I see two possible problems:
- This will fail during DMA as you will only see the DMA reads on the external bus. However, for classical Game Boy games this should not be a big problem. Most just keep the CPU in an idle loop during DMA and the few that actually use the time tend to read values from HRAM in order to write to some registers. The CPU barely ahs access to anything during DMA, so I would not expect anything relevant to your project. However, if you also want to do this for Game Boy Color games, things might be different. The GBC has more  DMA operations which might make it more relevant to use the DMA time. But since the Interceptor does support color mode, I did not look into color games much and cannot say if it might be a common problem there.
- I do not think that the data is visible even without DMA. Unfortunately, it has been a while since I worked on these details for the GB Interceptor, so I am not 100% sure, but if I remember correctly, the address lines on the external bus tend to match the internal bus activities, but the data lines don't do anything. There is no need to drive the external lines during internal access, so seeing anything there is a byproduct of an electrical design that does not care what's on the bus as long as the highest address bit tells the module to ignore it. In fact, at some point I used the address line state during internal access instead of the address pointer of my register clone as I thought it should be more reliable, but then I removed this as it turned out that one of the Game Boys (I think it's the DMGs) does not always set these lines properly. So, at the very least, if internal RAM data is visible on the external bus at all, you will have to check this for different Game Boy models.

[This page](http://iceboy.a-singer.de/doc/mem_patterns.html) also suggests that the data pins are not driven during internal access (except for a glitch). It says that they are only driven if the read line is high, regardless of CS and write, and that is not the case for internal access. So, to me it looks like the only way is to do the whole GB Interceptor thing and follow CPU instructions to get RAM data. (Well, you can ignore VRAM and do not have to emulate the PPU.)

odelot, 2025-04-07:
Thank you for the ultra-fast reply.

I did this Game Boy PoC during one day, using all I have already done for NES. I didn't logic analyzed the signals or anything. BUT I read this page you shared and could test and verify the behavior described: I could capture the HRAM address but not the value (or not the value I was expecting).

After I finish my NES project, I will try to debug your approach of intercepting the CPU instructions to interpret the RAM writes. (this sound like using a bazooka to kill a fly just because of 127 bytes of HRAM, but maybe it will be the only way)

Thank you very much for your time! If you want to join in during the effort to make a GB adapter, please let me know. 

I am not sure exactly how you rolled your project out, but it is being hard to people understand the open-source origins of mine. I often cite your project (and open source cart reader) as an example of how a project like mine can reach the potential users in the future. 

I will close the issue, thanks again.

odelot, 2025-04-07:
I am already just looking for memory addresses that are related to achievements. It is usually ~30 addresses. I transfer the data from the bus using DMA to a ping pong buffer, but almost like if it was a logic analyser. When I detect the value is stable, I test if the address is an address of interest and send it to the other core, that process all the RetroAchievements logic (using their lib, the same the emulator uses).

I am also pushing pico to its limit in terms of clock (started with 250Mhz, but now I dropped to 200Mhz) and RAM (unfortunately I need ~32kb for receiving the achievement list from RA - I could optimize it a lit bit transforming all my buffers in dynamic memory and just allocate them as needed - since this huge json is just important in the beginning of the process, before I start monitoring the BUS).

I will study your code to understand how it handles the writes on HRAM and WRAM. It is not clear to me yet if you simulate the CPU and its registers to know the value it would write to the memory. 

about the opensource part, it was not a question, it was just me venting out lol.... don't worry about that.